polyend/TrackerBetaTesting #1180: Mixer clipping indication

DataGreed, 2021-08-05:
Currently the tracker mixer has a very confusing indication for clipping sound as it is shown with color. The level bars with shifting colours look beautiful, but they don't provide enough feedback to quickly identify if the sound is clipping.

Sometimes the sound audibly distorts when the bar is orange, sometimes when it's red, so it's hard to tell.

I propose to add some visual indication for clipping.

E.g. it can be done by adding a horizontal line at 0 db point, so we'll know that the sound is clipping if the bar goes above that point (like in Ableton's mixer).

The other way is to add some kind of a visual LED-like indication, a flashing light when the sound is clipping, like OP-1 does it.

DataGreed, 2021-08-05:
BTW, the other unpleasant effect of current mixer implementation regarding clipping is that I usually set the levels lower to avoid distortion/clipping and when I use Preview feature in Sample Loader I am usually defeaned by a loud sound of the preview.

bignutemporium1, 2021-08-13:
Yuppp, I have blasted my ear drums a few times going into Sample Loader