polyend/TrackerBetaTesting #1182: Question to the community; Best firmware for avoiding internal noise + MIDI out

bignutemporium1, 2021-08-06:
What is your preferred firmware to use primarily for MIDI out as well as no internal noise?

tfspsound, 2021-08-07:
Hey bignut

If you are exclusively using midi and getting lots of noise do you need to plug anything in the output?
Also make sure the line in input is set to off as that can create some noise. Additionally the limiter can boost internal noise so you might want to check that. 
Not really noticed much of a difference in the internal noise between each OS it seems to have been fairly consistent. 
Hope this helps.

bignutemporium1, 2021-08-09:
Hey tfspsound,

Thank you for your response.

I do require output from the tracker, I sequence my x0xb0x with the tracker and then use the tracker for breaks + pads however I will just try without any output to see if this eliminates the internal noise.

The internal noise is not consistent, sometimes when I power on, there is a distinct electronic noise in the mix but if I turn off then back on, the noise is gone.

Checked for the line in and it is set to off, good tip though as well as the limiter; I will remember for this in the future.

Currently I'm on 1.3.1 but I will update to the 1.4b3 firmware, I was just worried about this particular update as I seen other users raise an issue where the internal audio comes out of sync.

tfspsound, 2021-08-09:
No worries, interesting that you have noticed the noise go away when you reset it. 
Im not entirely sure but I think there may be some issue with the limiter where if you set it to a channel and then turn it off it doesn’t seem to register that it is off. I’ll have to check on this again. 
With regards to midi stability I’ll have to bow out on that score I haven’t been sequencing external gear with the tracker since I upgraded to 1.4b1. Hope you find a workaround for the time being.

bignutemporium1, 2021-08-10:
I'll hold out for the next final release, thanks for your help tfspsound

closed 2021-08-10 by bignutemporium1