polyend/TrackerBetaTesting #1186: Song Mode : Colour code song blocks using Note, Instrument, Fx1, Fx2 keys

Aisjam, 2021-08-11:
## idea!
While "rec" in song mode. Selecting a single or group of blocks and pressing one of the 4 coloured keys to change the block to that colour, if you press the same colour again it goes back to white. 

## Why
I usually fill all 8 tracks in song mode so most of my songs look like a big square. This feature would to help make it easier to track where i am at with a song, also could colour code parts that are Intense with Fx1 Magenta and chill parts with fx2 blue. There is a similar feature on LSDj where tripple B will highlight a block, I use this a lot to manage where I am in a track or Key Points.  

I was pressing buttons and noticed they didn't do anything in the song mode so could be an easy way to add readability to the song mode. also would be an unexpected way to add pixel art :).

anemlife, 2021-08-11:
Cool idea! We definitely need something to make the song mode easier to visualize, at the moment it gets really difficult to work on long tracks

(Also a white skin would be cool, difficult to see the screen with the sun reflections :'( but that's another point ahah)

LetsSeeIfWeSolveThis, 2021-08-18:
''I usually fill all 8 tracks in song mode so most of my songs look like a big square.'' Haha, I feel you.

dramaone, 2021-08-23:
This is important! I mentioned in another thread and asked for either the ability to add text or coloured blocks.
When you have many variations to drum breaks it really ruins the flow of arrangement

dias236, 2021-09-01:
It would be Nice and Easier to use the song mode with color coded tracks, where patterns have distinct tones of their track color.

dias236, 2021-09-01:
It also would be super dope to have a small number identifying the pattern on the rectangle

AstralScare, 2021-09-01:
This would be amazing as color coding is something I'm used to from using Live

TwistedGenetics, 2021-11-17:
I been wanting this for a long time

colonel-blimp, 2022-01-21:
> While "rec" in song mode. Selecting a single or group of blocks and pressing one of the 4 coloured keys to change the block to that colour, if you press the same colour again it goes back to white.

This is a great suggestion.  I'm working with larger projects now, and I can see how this would be a godsend—right now, a large song using 8 tracks looks like a solid wall of white bricks!

I especially like the UI suggestion of using the colored keys (that currently have no function in song mode) to pick the colors.

dramaone, 2022-01-21:
solid wall of white bricks - exactly !

colonel-blimp, 2022-01-31:
The @Aisjam colour keys trilogy (so far):
1. #1186
2. #1215
3. #1387