polyend/TrackerBetaTesting #1194: Sound Editor Extract from sample

tfspsound, 2021-08-18:
We can crop a large sample down to a much smaller sample but what can we do when we have already set up slices within that large sample and memory is nearly maxed out? 

For example you have been working on a song for a while and an idea comes to you which relies on pitch shifting a small snippet of a larger sliced sample furthermore the snippet isn’t at the start of a slice so it isn’t possible to just play the slice and use Tune FX for pitch and Cut to stop the slice playing back fully. 

Currently we would either have to add a new slice at the point needed which would effect the subsequent slices in our song and become a real pain to fix. Or, destructively crop the sample and then import the sample again, set up the slices again and change the instrument value to the new locations value, or re-record the part/s

Am I missing an easier way to do this or would an option to Extract from a sample prove a useful addition?

Anexium, 2021-08-19:
This is very similar to my request #1170, expect yours is for specific regions whereas mine was for the individual slices.  Fundamentally it's the same thing, just with a slightly different way of choosing the regions.

tfspsound, 2021-09-05:
Just to elaborate on this slightly. It differs from converting slices by allowing us to ignore parts of the sample completely, saving memory. I like the convert slices idea but you would always require the same amount of memory as the sample that is being converted unless there was the option to replace the original. This suggestion would be more like the way you can edit a recording on a digitakt for anyone who has used one, but being able to do it whenever needed.

Jacknametrouble, 2021-12-18:
Cropping sample from the SD card in the Sample loaded is available since 1.3 firmware.

closed 2021-12-18 by Jacknametrouble