polyend/TrackerBetaTesting #1210: [1.4.0b] Delay does not sync properly when clock in = midi in jack

antf4rm, 2021-08-31:
### Steps to reproduce

Use midi in jack to sync to external clock 
use delay on a drum pattern and set delay to sync
repeats are not properly synced as they are with internal clock.

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")
always

* Does this bug happen in previous firmware versions?
yes

* Does this bug happen in every project?
yes

Jacknametrouble, 2021-09-04:
@ambv

OmskInfo, 2022-01-14:
The clicks are probably due to the delay constantly adjusting its time as it gets jittery tempo from the MIDI clock. I remember the Elektron Machinedrum doing this, too. Probably needs some kind of low-pass filter to ignore the jitter and only change the delay time when a new average tempo is detected.

sebpmtl, 2022-01-14:
yeah, anything that's clocked is affected by the jitter. It sucks. Even the midi clock out jitters.

Midisplitter, 2022-06-06:
Any updates yet on this bug? 

I'm trying to use the Tracker with another sampler, but if I use the Tracker as main brain, it stops my clock each time I change project. And if I try to use the other sampler as clock, all the work I have done on the Tracker with delay becomes unusable.