polyend/TrackerBetaTesting #1211: [1.4.0] Delay and reverb not applied to low level line input

PunyFlops, 2021-09-01:
Actions leading to the observed behaviour:

Effects send set @ 0dB. Effects return set @ 0dB. Input mode is set to “Stereo”. 
Apply low level signal to line input.

Behaviour:

The signal does not register with the effects (regardless of the fx send level). Increasing line in signal level will at som point lead to the effects being applied as expected. Audible and with a corresponding reading on the return level meter.

In order for this to happen, the signal on line in has to be below a threshold, that seems to be well defined, i.e, it is not gradual but on/off.

The behaviour is reproducible and persistent across projects and power cycle.

trutypesounds, 2021-09-01:
been mentioned previously here #1109 for example

ChielZ, 2021-12-24:
I have run in to this issue as well. I would like to add that on the delay, this might actually be a feature rather than a bug. The higher level required to trigger the delay results in an envelope (slower attack) on the delay of short notes, which creates an interesting sound. 

For me, the ideal fix for this issue would be to make the threshold adjustable, ie to add a slider to the delay and reverb pages making this parameter available within the tracker.

PunyFlops, 2021-12-24:
I agree, I had the same thought.