polyend/TrackerBetaTesting #1212: [1.4.0] Live Recording not working properly / stops working after deleting sequence and other strange circumstances

bndofc, 2021-09-01:
the live recording function currently has some issues that have been present since 1.3.1. when you have only one track armed, fast playing in note mode (in the instrument mode it works fine) is not possible at all. i assume this is due to the way it handles OFF messages. it would be nice if there was an option in the config menu to disable OFF messages entirely, so it wouldn't choke on the off messages, if that's what's causing the issue.

the other problem with this function is that it'll stop working entirely under strange circumstances, like swapping which tracks are armed between recording sequences after a while, but i think i've found a consistent way of making it stop working instantly; here are the steps

https://user-images.githubusercontent.com/80283211/131727067-25f04a08-fc63-454b-a6a3-d51660d6b052.mov

1. have only 1 track armed
2. enable live record
3. input a sequence
4. exit out of record mode (so note lane is green)
5. engage live recording once again
6. delete sequence exactly when the playhead is on the first step
7. you will have a hanging note and live recording function stops working.

only way of fixing it is by restarting transport. other instances of this happening have been reported in #938 #1051 #956

bambrose, 2021-09-01:
Agree on both points
Fast recording is not possible
Tracks become "disabled" although this can usually be resolved by stop start

adamvlasak, 2021-09-01:
I can confirm this bug.

Jacknametrouble, 2021-09-04:
@ambv

PITmast, 2021-09-08:
This is because you deleted OFF commands.
You have to erase everything after OFF command and before next note to avoid hanging note.

I reported it months ago. Too pity it’s still there.

Regards

visualearner, 2021-12-02:
I’m still experiencing both of these these things as well along with bad pad sensitivity. For me it makes it difficult to even use the unit at all at times. With a bug this severe I wonder why it’s not been addressed and how many people experience it.

closed 2023-04-21 by MiroPoly