polyend/TrackerBetaTesting #1217: [1.4.0] When changing pattern immediatly using the wheel MIDI sync is lost sometime

IohannRabeson, 2021-09-04:
#1216 

It happens only if you have 2 patterns with diffent lenght.
For my test my first pattern has a length of 64 and the second one has a length of 8.

## Steps to reproduce

The first pattern has a length of 64 and the second one has a length of 8.
Use an external MIDI clock, connected through MIDI in jack to the Tracker (please configure the Tracker to receive clock and transport from external gear).
Start play with the external clock, the Tracker follows as it should. Press pattern key and change the current pattern to the previous one, do not release the pattern key, and change the current pattern to get back to the previous one, then repeat, go again backward then forward.
You will quickly notice (after few backs and forths, 3 or 4 usually) the clock is lost at some point. You have to stop the external clock, then restart it. 

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")
If you do few back and forth you will get it.

* Does this bug happen in previous firmware versions?
It happens also with 1.4.0b4.

* Does this bug happen in every project?
Yes

Jacknametrouble, 2021-09-04:
@ambv

antf4rm, 2021-09-07:
yeah, seems like faster tempos have this issue more often.