polyend/TrackerBetaTesting #1218: [1.4.0] Crashes in Sample Playback mode when previewing beat slices

tfspsound, 2021-09-05:
I had just edited a large sample into beat slices and was quickly playing through them in no particular order while still on the Sample Playback screen when the tracker froze completely requiring the power to be pulled from the back of the unit. 
I may have pressed 2 buttons simultaneously to trigger this bug or there may have been an overlap between the previous slice playing out and the new slice being triggered. 
I had this happen before in firmware 1.4.0b1 but not before then, I haven’t tested it on an older firmware.

Jacknametrouble, 2021-09-05:
@ambv

tfspsound, 2022-01-05:
I can’t reproduce this anymore. I added a slice point in the sample and the issue appears to have gone.

closed 2022-01-11 by MortalJS