polyend/TrackerBetaTesting #1219: [Firmware version 1.4.0] Tracker self installed old firmware

distorson, 2021-09-05:
Hi, 
I was on firmware 1.4.0. The tracker somehow started to install the old 1.3.0 firmware **by itself**. Very weird. Than Tracker got stuck after 100%. The screen froze on the 100% bar of the firmware update. I pressed the on / off switch and without rebooting the tracker showed up the pattern I was just working on. But a few of the sound parameters where not the same anymore. The volume settings on some of the tracks where lower than before and also the envelope of the pan settings on at least one track was different. 

When I reinstalled firmware 1.4.0 the track settings where back to normal.

### Steps to reproduce
I left the Tracker switch on showing the pattern screen, not playing. It was connected via USB to my computer (MacBook Pro MacOS 10.13.6). The screen of my computer was dark. After I pressed a key on the keyboard of my computer the Tracker started "self installing" 1.3.0. 
Maybe the problem is connected to the issue that my computer always tries to open Android File Transfer when I connect the Tracker to my computer?

### Other important information

* How often does this bug happen? So far it only happened once

* Does this bug happen in previous firmware versions? No

Cheers, 
Hendrik

dimatura, 2021-09-06:
You might find similar past tickets here. IIRC this is a hardware issue.

bambrose, 2021-09-06:
…. It was related to the reset button
When activated it installs a “random” firmware from the firmware folder on
the card
So remove all but the one you want and it will only install that.
IIRC there was some very “sensitive” reset buttons causing this .. might be
with checking it’s not trapped between the case and circuit board or
something?

Jacknametrouble, 2021-09-06:
May be a hardware issue, if this happens on its own report to https://help.polyend.com/submit_ticket

closed 2021-09-06 by Jacknametrouble