polyend/TrackerBetaTesting #1220: [1.4.0] Tracker jumps itself to pattern 0 sometime

IohannRabeson, 2021-09-06:
## Hi, sorry you had trouble!

### Steps to reproduce

Honestly I don't understand when its happen, seems pretty random.

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")
Its quite rare, but I really don't want thats happen to me on stage..

* Does this bug happen in previous firmware versions?
I think I never experienced it before butI'm not sure

* Does this bug happen in every project?
Not sure as well

Jacknametrouble, 2021-09-06:
@ambv

static-noise-PT, 2021-09-07:
I can conform this. 
Never happened before 1.4
Could not recreate it. From what I remember I was switching from pattern page to sample page and back and was at pattern 0 :D

antf4rm, 2021-09-07:
Also had this happen when making a new project. Starting making a pattern on pattern 0 without realizing it. Then I would hit play which would make the song play from pattern 1 and all my steps would disappear, When i realized i was on pattern 0 i was able to copy my program to pattern 1.