polyend/TrackerBetaTesting #1222: Highpass Filter Before Effects – Sound Enhancement Neeeded for Master Out

static-noise-PT, 2021-09-07:
Till now I mostly record my performances with the Tracker and never used the "rendering" function beside for resampling.
Due to this fact I'm carefully with panning, gain staging and filtering in order to get the best possible recording.
But I quite often run into the problem that low frequencies send to reverb/delay causing (naturally) some problem:
it can get muddy but I see the possible mess with the phase correlation as the bigger problem. When not consider this in before one can run into mono compatibility issues with the recording. Therefore I ask you to implement a simple 12db hp filter before the delay/reverb. For me this is the last essential piece missing to make the mixing capabilities of the Tracker almost perfect.

trutypesounds, 2021-09-07:
mentioned here #1172

static-noise-PT, 2021-09-07:
> 
> 
> mentioned here #1172

I saw this issues but was a bit lazy and did not read everything thoroughly, sorry for that :D 

But we need a HP filter! Polyend please. I know you can't change the fx code probably but I assume you can place a filter before the reverb/delay. Peace

wesleyhodgson, 2021-09-13:
I'd also love a filter on the delay/reverb feedback path, akin to what's found in the Digitakt/Digitone. It's obviously super helpful for mixing, as stated above, but it's also super helpful for creative purposes like getting dubby sounding delay lines.

closed 2022-05-17 by static-noise-PT