polyend/TrackerBetaTesting #1233: [1.4.0] Export Song Stems causing glitch on master wav file

trutypesounds, 2021-09-13:
Export song stems, listen to the master output wav file, various random audio glitches at the start of the file appears (sometimes its a loud glitch click, sometimes its a crackle).

p.s.
dear Polyend, this is getting really annoying with all those bugs all over the place =/ 

* How often does this bug happen? 
always

* Does this bug happen in previous firmware versions?
not sure

* Does this bug happen in every project?
yes

anemlife, 2021-09-14:
Same in previous versions! 

I always do another export of the master section in addition to tracks stems, it kind of works like this

clickysteve, 2021-09-26:
Confirming this is still an issue.

It's been a known bug for some time now. Is it possible for this to be added to the roadmap to fix?

[image]

clickysteve, 2021-09-26:
Noting this was reported here back in January: https://github.com/polyend/TrackerIssues/issues/697

MortalJS, 2021-11-09:
@ambv 
Bug confirmed. 
As in the description Export Song Stems causing glitch on master wav file.

trutypesounds, 2022-01-18:
@MortalJS @ambv bug is still here in 1.5 while using Export Song Stems - causing pop click on master file

closed 2022-03-19 by MortalJS

Jacknametrouble, 2022-04-29:
Fixed in https://github.com/polyend/TrackerBetaTesting/releases/tag/1.6.0b2