polyend/TrackerBetaTesting #1234: [1.4.0] Global instrument amplitude produces a click when turned all the way down

orbitfold, 2021-09-16:
## Hi, sorry you had trouble!

### Steps to reproduce

Take a short percussive sample and reduce the amplitude in instrument parameters to the lowest value. It will still produce a click during playback of that instrument. Which seems to mean that volume amplitude is applied with a delay. It does not matter if you apply a volume envelope or not.

### Attachments and links

Often the best way to show the problem is to record a short video,
attach the Tracker project you were working on, and provide links.

Pro-tip: You can drag and drop images and `.zip` files directly to the edit
box when writing your bug reports. Videos are too big as attachments, you
will have to use an external service for this and provide us links. Please
don't use services that delete files after short periods of time.

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")
So far always
* Does this bug happen in previous firmware versions?
No
* Does this bug happen in every project?
I don't know

k4ot, 2021-09-17:
i had this issue with a tom sample played at low volume fx, it always comes out clicky

antf4rm, 2021-09-17:
yep, ran into this yesterday. Turned a sample all the way down but it still had an attack sound left behind.

MortalJS, 2022-01-17:
Hi @orbitfold,

Can you uplaoad some simple project with no more than 3 samples to fast reproduce problem?

MortalJS, 2022-02-20:
Closed. No response

closed 2022-02-20 by MortalJS