polyend/TrackerBetaTesting #1235: A convenient way to switch scales mode for pads on and off

orbitfold, 2021-09-16:
## Hi, we're excited to hear your idea!

Before you start, did you check whether your feature was already requested?
Searching might save you quite a bit of time here.

### A new feature

I really like the possibility to set different scales for the pads. However I think it would be useful to be able to quickly switch it back and forth from "c chromatic". Now I have to go to the settings for that. Mainly this is to easily be able to switch from "drum" to "melody/harmony" modes.

mwmwmw, 2021-09-16:
This and being able to transpose the pads without having to go into the config.

Shift Note Up Arrow / Down Arrow for transpose (+/- octave?)
Shift Note Left / Right for scale mode (cycles through?)

antf4rm, 2021-09-23:
Yes! Navigating to the Config menu for keyboard changes is a huge flow killer.