polyend/TrackerBetaTesting #1236: [1.4.0] Pattern length stuck when in perform/song mode

keith-macleod, 2021-09-17:
Steps to reproduce:
There are lots of ways to make this occur, but this is pretty direct and obvious:
- Create 2 patterns of different length
- Song mode: create a song alternating these 2 patterns
- Play the song
- Switch to perform mode

Whichever pattern was playing when the switch to perform was done will now be the pattern length in perform mode. If switched while on the longer pattern, there will be a gap after the shorter pattern. If switched on the shorter pattern, the longer pattern will be truncated.

Switching back to song mode will resume normal pattern length behavior, but behavior will again be altered when switching back to to perform mode, once again depending on which pattern was playing when the switch occurred.

I know that this goes back to at least 1.4.0b4, since I was running that when I came across this.

MortalJS, 2021-11-09:
@ambv 
Bug confirmed.
As in the description Pattern length stuck when in perform/song mode.

tfspsound, 2022-01-05:
Can confirm this is still present in 1.5.0b2

closed 2022-03-19 by MortalJS