polyend/TrackerBetaTesting #1238: allow FX without a instument/note trigger

k4ot, 2021-09-17:
allow FX without having an instrument/note trigger.

i think we would gain 2 things by adding this: 

-first you could place a bend, or chance fx, or whatever fx you can think of, acting on the last trigger, like write a pitch bend after xxxx steps without the need to retrigger the instrument. 
-not sure about this one, but it could de facto greatly increase mod playback which, well, let's not be salty, is not that accurate in its current state. That is standard software tracker behaviour, I can't wait it gets his way to the PT.

This should be doable whatever the FX is.

antf4rm, 2021-09-23:
This already works. It def works with J's (Slide down).

theorize999, 2021-09-24:
Hasn't this been a thing for a long time?