polyend/TrackerBetaTesting #1240: faster envelopes

Peterbing, 2021-09-18:
Hello! Loving the Tracker for slicing up drum breaks and re-sampling full tracks! 

However — and I’m surprised this hasn’t already been mentioned, although I did search — imo the envelopes (particularly the attack phase of the volume envelope) are not fast enough to get the best results when slicing beats. 

With an `Attack` of `0.000s`, one risks clicking at the beginning of each slice (which is fine if that’s the sound you are going for), but a single step forward on the encoder yields an `Attack` of `0.010s`, or 10 ms. 

imo, 10 milliseconds is too slow to remove a (bad-sounding) click at the beginning of a drum slice without also losing some of the (good-sounding) “snap” of the drums. At 10 ms, very tightly-sliced drum slices sound “puffy”, which is not ideal.

It’s odd because there’s another digit in the “`0.010s`” visual display, but it doesn’t seem actually possible to get values under 10 ms. imo the granularity should be 1 ms here, assuming the envelopes are linear.

For reference the default fade-in, -out, and crossfade in Ableton Live is 4 ms, I believe. 

Thanks for the fun product!

Peter

theorize999, 2021-09-19:
I agree, I usually shoot for 4ms when doing something that needs a very fast attack time... this usually holds true when doing resonant bass sounds as well. Really fast envelopes use more resources though because it needs to sample at a higher frequency... if I understand it correctly. At this point I think most enhancements to the tracker will come at a cost. I think it's pushing technical limits. I've said it before and I'll re-iterate here too, I think the best way to improve the tracker and keep as many people happy is to build in some modularity: not every resource allocated to instruments is going to be used in all cases, like in your request maybe we could have faster envelopes by switching off one of the envelopes, or maybe having a different envelope mode, to be honest I'm not sure if having more or less stages helps with this but it's an idea. I'd like to see the ability to have morphing wavetables in certain instances and would be willing to give up quite a bit in those instances, for example if i want to have a morphing bass wavetable I don't necessarily need a filter, I make wavetables and can just design one that sounds the way I want without the need for a filter if the wavetable sounds good. I think a lot can be improved with no extra resources taken by just toggling the way a particular feature works. I'm ok with trade-offs, but know very little about the super technical aspects of how this works so I may be way far missing the mark here.

77377, 2021-09-21:
agreed

antf4rm, 2021-09-23:
Yes! Need more precise times on short attacks.

tomasoma, 2021-12-14:
agreed

whannah1, 2021-12-19:
I would love if the shift key would allow us to change the envelope time in increments of 0.001 sec.