polyend/TrackerBetaTesting #1241: “smart” envelope hold phase for drums

Peterbing, 2021-09-18:
Hello once again, and thanks for the cool machine, and for taking suggestions here!

One of my favorite features of the older MPCs (2000xl, 1k, MPC 500, etc.) is how the envelopes work when samples are in “one-shot” mode. I use this mode for both traditional “one-shot” drum samples as well as slices of breaks of entire other songs. 

In this mode, the user has only controls for attack and decay. The attack works as expected, [de-clicking](https://github.com/polyend/TrackerIssues/issues/1240) or softening the onset of the sample. 

However, the “decay” does not work like a normal decay. Instead, the value entered as decay really adjusts the fade-_out_ of the tail _end_ of the sample. For chopping, this works much better than an ADSR envelope because the user-selected envelope values are anchored to the slice points (which the user has already selected very carefully), not to the gate length of the note (which is a much coarser measure).

One way to fake this is with some math and an AHD envelope: if we imagine that the user has sampled 500 ms of audio and has selected 2 ms of attack and 48 ms of “decay”, our final envelope values become: A = 2, H = 450, D = 48. There is no sustain phase of the envelope.

This feature is — imo — why chopped breaks and music beds can be “re-sequenced” with more natural-sounding results on an MPC than on the Tracker: the combination of exact slice points plus fade ins and -outs that are always anchored exactly to those points gives repeatable results with no gate length fiddling, no matter how much one wants to shuffle the beat around.

Another way to see this same effect is to slice a break on the Tracker and then try to play it live like a drummer (or, indeed, like someone playing an MPC live) using the grid buttons. The fact that there is a sustain phase to the envelope makes it basically impossible to play the grid buttons in a drum-like way.

I understand this is potentially a rather large feature request (an entirely new envelope mode that has to “know” the length of the underlying audio?) but it’s top of my personal list for turning the Tracker into the ultimate slicing machine! So, I had to ask 😄

antf4rm, 2021-09-23:
Agreed. This would improve the live input.

Peterbing, 2021-09-28:
> Agreed. This would improve the live input.
> 
> However, I believe you can do this by turning release and sustain all the way down and only using the decay portion of the envelope? But, that might not be exactly the sound you are trying to achieve.

It’s not exactly the same. On the MPC there are two modes. One is called something like “start”, and it is similar to an AD envelope, yes. However, the envelope runs to completion for every note, no matter how short. So you could (I believe) do this on the tracker, but D and R would both have to be set to the same number. 

The “end” mode is much cooler. It anchors the “decay” time to the end of the sample audio, no matter how long the input trigger is. So “attack” is effectively “fade in”, and “decay” is “fade out”. This is by far the best kind of envelope I’ve ever used for chopped drum samples.