polyend/TrackerBetaTesting #1243: [1.4.0] MIDI bugs with perfom FX

keith-macleod, 2021-09-19:
Two bugs, maybe related. First one is that step repeat has no effect on midi notes unless tune effect is also active, maybe others too, I can't claim to have done an exhaustive test.
Second is that when using note repeat and tune FX together, it seems to be a near certainty that a note will get stuck on on the external instrument, as thought note off messages were being dropped, or maybe sent out of order.

Steps to reproduce the first:
- Sequence an external synth on any track.
- On perform page try to apply step repeater to that track.
- Result: no effect.
- Expected: step repeat effect.

Steps to reproduce the second:
- Start as above, make sure that the synth patch has sustain (to hear the stuck note)
- Also apply tune fx at the same time as step repeat. Step repeat now works.
- Remove both effects/stop the sequencer. There will be notes stuck on on the external instrument.

This seems to happen very nearly 100% of the time, although anecdotally I thought I had done this previously without stuck note, but I may have just been using a patch with no sustain.

There's another weird level of interaction between these 2 effects on midi instruments, it seems like longer beat repeats (i.e. 2 and above) may not actually repeat beats, but they seem to add an extra pitch shift (only when tune effect is active, when it isn't, there is no effect, as described above.)

MortalJS, 2021-11-23:
Hi @keith-macleod

I am trying to reproduce the problem you have described. In my case when I follow exactly your description step repeater works on external midi device (in performance mode) regardless if I use Tune FX for it or not.

MortalJS, 2022-01-11:
Closure due to lack of feedback and no confirmation on another device.

closed 2022-01-11 by MortalJS