polyend/TrackerBetaTesting #1244: 1.4.0 granular/wt LFO bug and sample position and step repeater crash freeze in perform mode

anton10w, 2021-09-19:
### Steps to reproduce

All tracks with midi notes and sequences etc. up to 6 patterns, fx sends to reverbs etc, sequence running at 169 bpm.
When in performance mode after using "sample position" and or "Step repeater" the tracker crashed I was unable to use the power button to hard reset it, crash while sound is still coming out, only way to stop the bug was to pull the power usb cable, I have attached a video showing this.

https://user-images.githubusercontent.com/90134130/133926131-95c5fbb3-80ac-421d-9487-3ce30bc2293f.mov

Afterwards with the same song, If I engage the granular/wt LFo speed , the effect stays on, even if I quit  that particular effect  and use something else. 

* How often does this bug happen? (every time, often, "it only happened once")

step repeater and sample position in performance mode happened once.

Granular/wt lfo happens often.

* Does this bug happen in previous firmware versions?

Not sure

*** Does this bug happen in every project?**

RE : granular/wt LFO 

Managed to recreate on another project, the only way to get out of is to go back to pattern mode, stop the tracker and press play again.

project, 3 tracks, i notes ,drums , kiks ,rollin hats etc  in **_pattern mode_** .
9 effects with pad effect trigger engaged all tracks enabled, rec mode on  in **_perform mode_**

project @ 130 bpm

RE : sample position and step Repeater

I think I changed the settings as a workaround you can look at the video an try to recreate this

All the best

MortalJS, 2021-11-09:
Hi @anton10w  I am trying to recreate the situation seen in your video. Unfortunately without any effect. My device did not crash in this configuration. I am wondering if this is somehow related to the sample used.  Could you please attach a simplified project with sample which causes this situation.

MortalJS, 2022-01-11:
Closure due to lack of feedback and no confirmation on another device.

closed 2022-01-11 by MortalJS