polyend/TrackerBetaTesting #1245: [1.4.0] Roll FX acts differently in pattern / song mode

fulhjort, 2021-09-19:
If you put a Roll FX anywhere in a pattern and play it, it will roll forever (or until you stop it somehow). But if you play it in Song mode the Roll will stop as you come to the next pattern, making it impossible to create a longer roll.

It's been like this since 1.3.0.

MortalJS, 2021-11-25:
To create a longer roll you can extend the pattern length to 128.

fulhjort, 2021-11-25:
You're misunderstanding me.

Create a 128-step pattern, place any instrument with a RN 1 on step 105 in Track 1. Play pattern. The roll will keep going as the pattern repeats, until it gets to step 105.
Create a second 128-step pattern, place any instrument on step 12 in Track 2. Play as song, once pattern 1 is over and pattern 2 starts, the roll will stop.

That's a bug, not a feature.

MortalJS, 2021-11-28:
Hi @fulhjort .
Thanks for additional explanation. 
I classify the report as a bug.

fulhjort, 2022-07-01:
This is still a bug in 1.6.0.

closed 2023-04-27 by borsukpolyend