polyend/TrackerBetaTesting #1247: [1.4.0] Tuning rolls (Rn and RN) work wrong with slices

anemlife, 2021-09-20:
When working with slices the Rn RN commands move slices up/down while it should move the tuning up/down.

MortalJS, 2021-11-09:
Hi @anemlife I'm trying to reproduce your submission not quite understanding what the problem is though. Can you attach more information?

anemlife, 2021-11-09:
Hello @MortalJS 
I did a quick video showing the bug.

To explain it better: note rolls (Rn and RN) work fine with any instruments type apart from sliced ones. 
Because the note roll FX kind of jump up/down pad notes. So when using it with slices it jump to the next/previous slice/pad while it should keep on playing the same slice while changing the note/tuning. (otherwise you can't really use it as expected)

https://youtu.be/9WGyRcj20Wc

Let me know if you need any more info.

MortalJS, 2021-11-22:
Hello @anemlife
Your video is enough for me to approve the bug. Thanks for describing the problem in detail.

@ambv  Bug confirmed.

piotr3500, 2021-12-29:
Hi @anemlife, I've checked your issue once again.
It can be easily fixed once you set your Sample Playback to Slice (not Beat Slice).
We designed the system in such a way, that you cannot change notes when you are using Beat Slice.

closed 2021-12-29 by piotr3500

anemlife, 2021-12-29:
Thanks for your answer @piotr3500 

That means we have to "spend" one fx lane to choose the right slice + another to do the "tuning roll" => somehow limiting as you have to use the two lanes for this... But I understand that's the way it is designed. 
Hope you can "rework" this part one day! Thanks for making this clear :)