polyend/TrackerBetaTesting #1249: [1.4.0] Copy/paste from song mode not reliable

anemlife, 2021-09-20:
If you copy/paste multiple patterns at once it behaves strange (sometimes it works sometimes not). Same if you press undo it doesn't go to the last state, it only undo certains patterns and make you lose other ones. (also works randomly)

MortalJS, 2021-11-09:
Hi @anemlife. I am trying to replicate your problem. Unfortunately without success. At the moment I can't forward it to the developers department.

anemlife, 2021-11-09:
Hi,

After a few try I understood what's happening so it's actually not a bug. 

If I copy/paste let's say 3 lanes at once and hit undo, it will first undo the third copied bar, then the second, then the first.
I thought it would undo the whole 3 lanes at once. But I got it now, closing this issue, thanks!

closed 2021-11-09 by anemlife