polyend/TrackerBetaTesting #1251: [1.4.0] Render selection bug

anemlife, 2021-09-20:
Sometimes it only render a few step and not the full steps selection. If you don't check what's been rendered you might lose some things. (also it happens randomly, did 3 tries with the same selection of 80 steps it finally worked the third time while at first it was only recording about 60 steps).

bambrose, 2021-09-20:
Yeah I've seen this behaviour too
It all relates to where in the grid the elements are.
What it likes is elements in column 1,2,3
If events are in 2,3 or 1,4 or something it doesn't work.
I'd been meaning to raise a ticket

orbitfold, 2021-09-20:
Yep, happens all the time

trutypesounds, 2021-09-26:
mentioned here #1162

i have the same issue (1.4.0 fw)

closed 2021-11-19 by ambv