polyend/TrackerBetaTesting #1252: Volume Ramp FX like the "D" effect from Impulse (schism) Tracker

antf4rm, 2021-09-22:
I'm a longtime Impulse Tracker user and I have to say i dearly miss the Volume slide FX.
so steps like this would quickly turn the volume down on every step. It was really nice for creating envelope release type fx to samples and great for giving beats a truncated tight feel as well.

d01 would ramp the volume down quickly.
d09 would ramp less quickly.

c-5-v64-d01
----v64-d01
----v64-d01
----v64-d01
----v64-d01

this would make a nice chopping effect on every step.

Please consider grabbing this classic FX type from Impulse Tracker. It's a must have!

bambrose, 2021-09-23:
Yes please. I Second this request

keith-macleod, 2021-09-26:
I'd like to see all the envelope parameters become targets for step and performance effects. Decay/release are definitely the most useful, however.