polyend/TrackerBetaTesting #1253: [1.4] Losing songs after 100 in the Projects folder

imehesz, 2021-09-23:
## Hi, sorry you had trouble!

### Steps to reproduce

Save more than 100 songs.

When you have more than 100 songs in the Projects folder, in the `Files` view, the tracker won't display them (only the first 100) 
To SD card stores the files, please see attachment.

### Attachments and links

[image]

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")

every time

* Does this bug happen in previous firmware versions?

yes (I just upgraded to the latest one)

* Does this bug happen in every project?

yes

manuel12374, 2021-09-23:
The same limitations applies to all file management across all folders: i’ve tested a d got the same limitations for example when reading the MOD folder: while in the SD card in the mod folder i have 550+ files, the file manager shows only the top 100 and the only way to load the 101 one is to delete one of the top 100. if the 100 girls is an hard limitations a good workaround would be at least to allow sub-folder file managmeent so that we could create sub-folders with max 100 files each.

trutypesounds, 2021-09-23:
mentioned many time here and there

Jacknametrouble, 2021-09-23:
This is mentioned in the manual.

closed 2021-09-23 by Jacknametrouble

imehesz, 2021-09-23:
Thanks for the reply  @Jacknametrouble 

> This is mentioned in the manual.

Yes it is, but it doesn't mean Polyend can't improve on it. And they really should. (I'll submit another ticket as a feature request)

thanks,