polyend/TrackerBetaTesting #1254: Split Left/Right Render output

SpaceCadet64, 2021-09-23:
Hi,

I would like to suggest the option to render a selection as two separate mono files for the left and right signals, rather than one stereo file.
This would mean that users could arrange a section that they want to condense down to a new instrument, and get a separate left and right sample, which could then be loaded onto two tracks and panned to maintain the stereo field in their original arrangement.
Currently, rendering a section outputs a single stereo sample, which will then converted into a mono audio file when loaded into the project, thus losing any panning applied in the original arrangement.

There exists a workaround at the moment, involving rendering a section to a stereo file, transferring it to an external audio workstation, splitting the stereo channels, then transferring it back to the Tracker. This is very cumbersome and time consuming and if possible, this new feature could greatly expand the compositional potential of the unit.

antf4rm, 2021-09-23:
Nice one!

therealyez, 2021-11-04:
I just wrote the same suggestion to support and was directed to github. I'm happy to see I'm not the only one hoping for this enhancement.

If it could be an option when rendering both tracks and master it would be perfect!

tomasoma, 2021-12-30:
i had the same issue but thought about a channel selection left or right when importing / loading stereo sample ... it would work for all tracks, master, selection.

therealyez, 2021-12-30:
> i had the same issue but thought about a channel selection left or right when importing / loading stereo sample ... it would work for all tracks, master, selection.

It's a good idea but it's actually the other way around of what was mentioned here. We need a way to render/export several tracks into left and right.
As it is now, we are not able to bounce in stereo. With your fine suggestion we would have the same problem unless we record master to another unit and import left/right. We can already do that if we use other equipment to do it

tomasoma, 2021-12-30:
> As it is now, we are not able to bounce in stereo.
I'm using [1.4.0] firmware and I double checked to be sure not being wrong and export pattern stems and export pattern master and render selection all create a stereo file respecting the pan of the instruments, so we are able to bounce in stereo.
In fact, I don't care about what solution will be prefered as well as we all can re-create the "stereo instruments" we want, but my solution is also optimizing the memory capacity of the SD card by not creating multiple files...

therealyez, 2021-12-30:
You're right, sry.
I love your suggestion!

SpaceCadet64, 2021-12-30:
That is a really interesting alternative way of achieving the same result.
Makes sense that using one file and simply selecting which channel to use would be more space efficient.
I guess it really just depends what is possible to implement.

tomasoma, 2022-01-03:
@SpaceCadet64 i tried the workaround you suggest, and i'm facing issues with the "stereo instrument" (a beat sliced, i took care to slice the two channel exactly the same) : most of the time that works fine but randomly accidents make it sounds very loud during a few steps ... did you experience the same thing ?