polyend/TrackerBetaTesting #1257: Add a hyphen between note and octave for non-sharp notes. "C-5" and "C#5"

antf4rm, 2021-09-23:
This way the octave column stays aligned.

C-5 01 --- ---
C#5 01 --- ---
A-4 01 --- ---
A#4 01 --- ---

Also the octave column should be selectable on its own so that changes can be made solely to the octaves of a track without changing any other data in the step.

theorize999, 2021-09-24:
freakin awesome idea

Anexium, 2021-09-27:
If you have a selection highlighted, holding the shift key while turning the wheel will move the notes up and down an octave at a time.  Although re-reading your post, are you thinking of using something like the 'Fill' function just on the octaves?

theorize999, 2021-09-28:
I think it’s purely an optics thing, to make it look neater, some trackers
put a hyphen character between the note and the octave, some people are
used to seeing it that way, just like hexidecimal. Protrekkr was like that,
it was my first tracker so I understand why someone would want that. It
does help a little when looking at the pattern as a whole as well in my
opinion, it helps it not all blur together

antf4rm, 2021-09-28:
First I was hoping to clean up the information visually by adding the hyphen. Most of my midi gear displays notes this way so it is also easier on the eyes from one piece of gear to another.

Second and less important, I was hoping to edit the octave column on its own which would probably include using the fill function as well just on octaves.  yep.