polyend/TrackerBetaTesting #1259: [1.4.0] Freezes Sporadically

clickysteve, 2021-09-26:
### Steps to reproduce

I have not been able to find a consistent reproducible step for this. However, the tracker freezes in this firmware version every so often. Mostly when playing larger samples at a different pitch for the first time, or when loading the file list. This was never experienced with previous firmware versions or betas.

DinkyMirage, 2021-09-27:
Finding my Tracker freezing much more often after updating to 1.4.0 as well. Having trouble establishing a pattern of the freezes happen. The last two times I was previewing a newly loaded sample on the pads and it froze when I switched screens. I am powered from a usb battery

clickysteve, 2021-09-27:
Just for more data points: I am using the same microSD card and PSU (wall) as before, but I never experienced this previously. I have bought a new (faster) microSD card to try and mitigate the issue, but it's strange that this issue wasn't present previously.

Gareth-Dawson, 2021-09-27:
Please find incident of freezing streamed here: https://youtu.be/mw59WeHqf3o?t=2982. It's not my clip so cannot give too much information,

mwmwmw, 2021-09-29:
I've seen other users with freezing issues. Try a different USB power supply. 

Apparently, some people have issues with the one provided by Polyend. 

I have always used a big USB battery and have never had any freezing issues.

k4ot, 2021-09-30:
Had this too. Nonstop bugs. Every 3 minute. Almost unusable PT. 
After ditching PT power block, +30 hours of playback and heavy tweaking, I had not one.single.freeze yet.
I dont know exactly what's wrong but changing the power block made a drastical change in stability for me.

NoMoreFreezesPls, 2021-10-14:
> ### Steps to reproduce
> I have not been able to find a consistent reproducible step for this. However, the tracker freezes in this firmware version every so often. Mostly when playing larger samples [...]

I have had this unit for exactly one week and have experienced about...... 12+ freezes. Love the unit, but now I know why I got it for only $420.00 on reverb now......very unstable firmware.

Jacknametrouble, 2021-11-19:
Guys, how are you powering your Tracker units?

trutypesounds, 2021-11-19:
> Guys, how exactly are you powering your Tracker units? Have you tried switching the power sources?

I had this once since june 2021, was powered via polyend adapter, just a short sample looped in sample playback, then pressed pattern - viola, tracker hanged with a high pitch noise, it took several reboots to back to norm.

DinkyMirage, 2021-11-19:
Was powering with the wall adaptor, computer and usb power brick with the
original cable. Tried a new usb-c cable that I had around and have had only
one or two freezes since, in all power scenarios.

Jacknametrouble, 2021-11-20:
I had some support cases where the original Polyend AC adapter was the issue. Certain "intelligent" power bricks with advanced power supplying systems can cause issues too. The simpler the power bank the better. If anyone's experiencing random freezes, please try using an alternative AC adapter.

demolen, 2021-12-01:
I have bought my device just yesterday and today for the first time I used it. It crashed after 10 minutes on the song mode with a high pitch sound. The shutdown button was also not responsive, so I had to detach the cable. By far, considering other comments there seems to be an adapter issue. Does polyend replaces faulty adapters if it is the case. I'm sure there should be even a law about this in EU regarding customer protection rights.

Jacknametrouble, 2021-12-02:
@demolen https://help.polyend.com/submit_ticket

pintsized, 2022-01-03:
Using the unit for the first time, experienced this freeze / crash within 30 minutes. Firmware 1.4.0, using the supplied cable attached to my Thinkpad's USB. The freeze occurred when in edit mode, changing the instrument parameter with the jog wheel. I had to pull the power to restart.

I tried the same project, aggressively spinning the wheel and changing various parameters, whilst physically manipulating the power cable, but was unable to reproduce.

Jacknametrouble, 2022-01-18:
@pintsized Have you tried how your tracker behaves on other powering sources and cables?

jcnrn, 2022-08-24:
I've been having this issue as well during normal operation. I have attempted to power via the Polyend cable, a USB A to C cable using an iPhone wall wart, a Motorola USB C fast charging cable, and a different USB A to C cable via my desktop computer and was unable to resolve the freezing. As can be seen in the video below, I have graphical stuttering while the audio remains fluid. This occurs not only during song playback but through general operation of the Tracker where it interrupts my control of the unit.

Video of Stuttering (please excuse the chiptune lol)
https://drive.google.com/file/d/1TqtPPpjB1-zgeGW1YGFa4AfS156NDru2/view?usp=sharing
Stuttering occurs at 1m23s and 3m08s

closed 2023-04-21 by MiroPoly