polyend/TrackerBetaTesting #1260: [1.4.0] Performance mode : clicky performance

trutypesounds, 2021-09-26:
## Hi, sorry you had trouble!

### Steps to reproduce

muting/unmuting causes clicks on short samples like kicks or sub bass, various filter fx makes different sample kinds click too, some other fx cause clicks sometimes. most times this is horrible, very rare they are fancy proper accidents, but mostly sound horrible. needs to be improved, must be smoother working.

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")
most times

* Does this bug happen in previous firmware versions?
yes 

* Does this bug happen in every project?
mostly yes

Jacknametrouble, 2021-11-20:
First of all, distortion and clicks in the Performance mode could be a result of very loud samples. Please try to lower down the sample volume in the Instrument Parameters menu.

closed 2021-11-20 by Jacknametrouble

trutypesounds, 2021-11-23:
> 

using loud samples in other hardware/software (tried alot, never was an issue in Renosie for example, or Digitakt or OP-1) aint causing clicks when muting/unmuting