polyend/TrackerBetaTesting #1269: [1.4.0] Line in will stay mono even when changed to stereo

sebpmtl, 2021-10-20:
Line in is inconsistent. It sometimes stay in mono and will produce weird clipping sounds. I don't know how to explain the frequency. It's like with all the bugs with this machine. It will sometimes do it, sometimes not. You can reflash the firmware and it will be fine or not...

MortalJS, 2021-11-09:
Hi @sebpmtl. I'm trying to reproduce your problem. I use an external 12 voice synthesizer (with stereo output) for this. Everything works according to the selected function. Can you describe what kind of signal you are putting into the line in?

Jacknametrouble, 2021-11-20:
https://help.polyend.com/797585-Can-audio-be-passed-through-from-line-in-straight-to-output-in-Tracker

closed 2021-11-20 by Jacknametrouble