polyend/TrackerBetaTesting #1279: Possibility to edit output CC table in MIDI Instrument Parameters

b8ck8tt, 2021-10-30:
Hello,

It seems like there is currently no possibility to edit the CC output table for MIDI Channel instruments. The default preset says that 0 is "Bank", 1 is "ModWh" and so on. That doesn't align with a lot of synths out there, so it would be really nice if users could edit this table, and potentially even save different versions as presets for different synths.

Thank you!

[image]

Jacknametrouble, 2021-11-03:
You are requesting an implemented feature. This is exactly the place where you choose the exact desired number for each of the accessible CCs.

closed 2021-11-03 by Jacknametrouble

b8ck8tt, 2021-11-03:
> You are requesting an implemented feature. This is exactly the place where you choose the exact desired number for each of the accessible CCs.

I was not talking about choosing the number. I was asking for implementing a way to change the text next to the numbers. If it's already possible, I don't think it's explained on how to do that in the manual.

For example, in this table 10 is "Pan". However, on my synth, it's not. And I'd like to create a present where "Pan" would correspond to the CC number that my synth is actually using. I hope I explained myself correctly this time around.