polyend/TrackerBetaTesting #1281: Combine multiple samples, copying one into another.

synthexpert, 2021-11-09:
### Combining 2 or more samples

It would be nice to combine multiple samples. For example the KOALA APP does this. It allows you to pick a sample and paste its contents into another one. So you have a new sample made of two layers.

I would love this for sound design purpose.

Thanks.

Anexium, 2021-11-10:
You could achieve this by using the 'render selection' function within the pattern menu's 'more' options.  Just drop your samples on adjacent tracks, highlight them and then hit the render selection option. It'd also give you more control over things like the relative volumes and filter settings for example.

Admittedly, this way wouldn't give you the sample-level accuracy that doing it through the sample editor would - but you could increase the tempo and I think that'd get you close enough.

synthexpert, 2021-11-12:
> You could achieve this by using the 'render selection' function within the pattern menu's 'more' options. Just drop your samples on adjacent tracks, highlight them and then hit the render selection [...]

Hello, thanks, is a good way (I know it of course...) to get a nice sound design byproduct but it's not the same. I was referring precisely to **sample merge.** Hope the tracker's hardware allows this workload of calculation.

tfspsound, 2021-12-06:
I made a very similar request #1055 but it got closed down fairly quickly. I’m still behind it.