polyend/TrackerBetaTesting #1286: Non-overwriting paste

jarkk0sakkinen, 2021-11-13:
# Problem

Sometimes you want to merge two tracks into one. For instance, you might have a track with an on-beat kick drum and a track with an off-beat bass. In order to save a track, you might want to merge these into a single track. Doing this manually takes a lot of time and is error prone.

# Solution

Have an alternative paste, which overwrites only cells, which do not have note information.

Anexium, 2021-11-16:
I like this idea, and think it'd come in really handy.

I'd also like to propose what is essentially the opposite idea - a way to spread the note data from one track over a user-specified number of tracks.  It's one of the handy little tricks that OctaMED on the Amiga used to do.

Imagine you've got a single track with your drums and percussions on.  As they play - depending on sample length - they'll cut each other off.  My proposal is that you could take the note information from that single track and spread it over a number of tracks.  That would then play but without the notes cutting each other off - which'd be great for a resampling, or just for putting other effects on in an easier manner.

Quick example (sorry if the formatting goes wonky) - 
```
Before
[track 1]    [track 2]
01 kick      -
02 hat       -
03 snare     -
04 hat       -

After
[track 1]    [track 2]
01 kick      -
-            02 hat
03 snare     -
-            04 hat
```

jarkkojs, 2021-11-22:
Yeah, agree that it should work in both ways! This is a major glitch in the workflow, not just a "nice to have" feature. You constantly have to do a lot of manual work when reorganizing a track, or when you want to resample bunch of stuff,

thumperjack, 2021-12-02:
+1, there is a fill option to only fill blank spaces, a paste option that only filled in blank spaces would be great to fill in spaces in a kick snare pattern with pasted parts of another track, or to merge two tracks together while keeping the basis of the first track unaltered

zikajk, 2021-12-28:
I just experienced this today -if you apply fill with density 60% multiple times you will get your track full. You must delete after every fill. 
To be honest - i think that paste should be one that does not replace your notes by empty spaces by default.