polyend/TrackerBetaTesting #1289: Previewing single cycle samples defaults to Forward Loop

SamNazar, 2021-11-22:
Previewing single cycle waveforms in the sample browser is currently difficult, as they all sound like extremely short ticks when only played as one cycle.

By default, the preview button for samples in the left panel should loop any samples that are shorter than 0.03 seconds.  This would make finding the right single-cycle waveform far more efficient, without affecting even the shortest one-shots.

bambrose, 2021-11-22:
Yes agree

MetabaronCZ, 2022-02-11:
I think there should be a small menu with possibility to one-shot or loop the sample. But most importantly volume slider for sample preview,

apzlo, 2022-06-22:
Absolutely. This is my #1  difficulty with using the tracker overall, single cycle waverforms are particularly important for trackers, it should be easier to preview while browsing them