polyend/TrackerBetaTesting #1290: [1.4.0b1] Fill for Slice effect in "From To" mode uses its range in a weird way

b8ck8tt, 2021-11-23:
1. Create a pattern with length of 10 (hex) / 16 (dec)
2. Fill every other step with a C5
3. Select FX1 column with Shift + Up Arrow, Press Fill
4. Where? - choose "Note", Fx Type - choose "Slice", Fill Type - choose "From-To"
5. From - choose "1", To - choose "8", press "Fill"

Observed Result: Slices are filled with double "1" slice and up to 7.
[image]

Expected Result: Slices are filled from 1 up to 8.

Additional Information:
If you choose From 1 To 9, it works in an expected way.
Sometimes From 1 To 8 works correctly, but if you fill with something else, and then try again from 1 to 8, it will fill with double 1s.
Same issue happens if you choose filling from 3 to 10, 4 to 11, etc.

MortalJS, 2021-11-25:
Bug confirmed. Transferred to developers.

closed 2021-12-18 by ambv

Jacknametrouble, 2022-01-15:
Fixed in firmware v1.5.0