polyend/TrackerBetaTesting #1295: note retrigger effect

thumperjack, 2021-12-01:
note retrigger effect as exists on octatrack, which still allows single note length subdivisions, but also (more importantly) allows longer than note length retriggers, so one note can retrigger one note a set number of times, at a set rate, or the rate can be bumped while it's retriggering for moving ratchets. it's also a quicker easy way to do long triplets or quintuplets without having to rethink the entire pattern length and grid spacing, and with a group of notes set off by retrigger, you can shuffle bump the whole group of notes by micro-timing just the one trigger note.

thumperjack, 2021-12-08:
hey you know what I didn't realize roll just kept firing away forever unless something stops it. all roll needs is the other resolutions, triplets/quintuplets/dotted, and also non-grid free time resolution

Jacknametrouble, 2021-12-19:
Thank you for your feedback.

closed 2021-12-19 by Jacknametrouble