polyend/TrackerBetaTesting #1296: 1.4.0 Constant and different noises/hums from all of the inputs

demolen, 2021-12-01:
My newly bought unit. A MK2 model with metal casing, Running Firmware 1.4.0 suffers from noise/hum from Line-in as well as Mic inputs. Line-in has a warmer/wider range hum which is quite loud, and the mic inputs have high frequency white-noise-sounding noise. I tried different cables with no success. The noise is there only if a cable is connected to the Line and Mic inputs. 

I second that the unity could be faulty on a hardware level, so I'm returning it to the retailer tomorrow. Maybe the QA could be more picky at the polyend side... Electron is a good example... the device also crashed once yesterday. I thought sequencers don't crash... I was mistaken, as it seems.

Thought to post it here, Just In case of the developer feedback or others with the same issue to track the process.

demolen, 2021-12-02:
update:

The big hum from the line-in disappears when I power the tracker from the USB port of my laptop. But still the white noise is there for all of the input ports.

demolen, 2021-12-02:
update: 
I can get rid of the line-in hum if I connect the USB cable first to a USB hub and then to the Adapter. The high frequency drone which is present in all the inputs is also gone in sample playback and preview mode. There could be a noise reduction algorithm or a gate or dynamic compression applied to the samples during the recording. Maybe someone from polyend side can confirm and elaborate why the noise is there during monitoring mode but not once the sample is recorded. The inputs are still too noisy for a 500+ bucks unit, and the company seems not to write a single word about it in their documentation. I would do that if I were them, so every user doesn't have to research on their own. And I'm sure they know.

demolen, 2021-12-03:
If it's not reproducible on your side, it means that my unit is faulty. Should I return it?

Jacknametrouble, 2021-12-03:
Does it happen when you turn off the audio input monitoring in the Master section?

demolen, 2021-12-03:
Are you indirectly confirming  that every unit shipped has noise while monitoring input?

demolen, 2021-12-03:
My conversation with the poly end support also ended to the same indirect affirmation. They suggested me to disable input monitoring. I returned the unit already. I’m still not trusting polyends business ethics though. Let’s see…

trutypesounds, 2021-12-03:
> My conversation with the poly end support also ended to the same indirect affirmation. They suggested me to disable input monitoring. I returned the unit already. I’m still not trusting polyends business ethics though. Let’s see…

yeah their ethics is shit and i expected a bit more effort from their side for such an interesting device, so glad i didnt picked it up on day 1 -  the initial state of firmware was terrible. its much better now, but all the weird quirks and bugs here and there... KURWA!

demolen, 2021-12-03:
@trutypesounds i just picked this up a couple of days ago and it already seems like an unfinished product, I can’t imagine what is used to be back then… It had crash in its first hour of duty, welcomed me with all kind of non documented noises, hurt my ears with that weird metronome sound and nearly got me a seizure for not being able to turn off the pattern follow. I also don’t know what That wavetable joke is about? Was someone asking for that? None of the included projects that I have looked through also have done anything with it. Feels like a waste of ram that could be invested elsewhere. Like polyphony…. Oh at least two voice Polyphony…

demolen, 2021-12-03:
But honestly „monovoice wavetable synthesis“ is something I have to laugh about

moshier1, 2021-12-06:
It's called a ground loop, you're using two different power sources causing the ground to be different and causing a hum. It has nothing to do with the device, just how you power it. Get a ground loop isolator.

demolen, 2021-12-06:
The hum was available even when the device was connected directly to the power outlet, monitored through a headphone, and the line-in was connected to my turned off monologue not plugged to the outlet. But thank you for your tip!

moshier1, 2021-12-12:
> The hum was available even when the device was connected directly to the power outlet, monitored through a headphone, and the line-in was connected to my turned off monologue not plugged to the outlet. But thank you for your tip!

were both devices powered by the same dc convertor tho?

demolen, 2021-12-12:
Yes. I also have tried other configurations to see if I can eliminate the
hum but I could not succeed.

demolen, 2021-12-17:
Update: Musikstore contacted me and they confirmed that the unit is faulty. They have compared it with a new model and the new model didn't have the noise. They said it will be sent back to Poland.