polyend/TrackerBetaTesting #1297: micro move negative values

thumperjack, 2021-12-02:
micro move negative values, so you can make a note fall early, at full min value the note would fall exactly on the previous beat, at max value the note would fall exactly on the next beat

tfspsound, 2021-12-02:
Hey
Wouldn’t this mean it would have to play the beat before it had been told to do so? Maybe micro move that could be longer than a step would work so for example if you set it to 120 then it would play 20 ticks into the next step and allow another micro move setting on that step eg 70. That way in one step you would have a trigger at 20 and 70 ticks.

Anexium, 2021-12-03:
Why not just move the note to the previous step and use the micro move to play it towards the end of the step?

thumperjack, 2021-12-03:
> Why not just move the note to the previous step and use the micro move to play it towards the end of the step?

that would take up the previous note. you need the penultimate beat and the downbeat both, but the down beat needs to fall early

mwmwmw, 2021-12-04:
I cannot speak for the code in the tracker, but I have written schedulers before and it would be pretty trivial to read ahead one tick to check for this effect.

If you can sacrifice a sample, you could probably render the two hits, then trim out the middle to get the second hit in just the right place.

tfspsound, 2021-12-07:
@mwmwmw Im guessing that’s how the digitakt works or something similar. I could be wrong but I just assumed that since we cannot assign negative values to micro move the tracker may not be able to do it. Being able to have 2 events trigger in the same slot when we want them (ie not using roll) would be very helpful.

mwmwmw, 2021-12-07:
@tfspsound From a programming standpoint, those values may be stored as an unsigned integer. 

Unsigned integers don't have negative values and so modifying the sequencer to support them would not be trivial.