polyend/TrackerBetaTesting #1300: 1.4.0 Every time I go into the menu to try to disable the LFO or change the speed, the unit freezes.

Jacknametrouble, 2021-12-04:
From Philip Meyer email: ntrm@inter-modal.com

"Hi I have a one-shot instrument and in the 2nd instrument parameters screen I've set the panning to be modulated by a random LFO at the fastest speed /128, which causes the unit to stop running and if the sequencer is running, it outputs a constant oscillator tone. The only way to stop it is to unplug the power cord and restart the unit. Every time I go into the menu to try to disable the LFO or change the speed, the unit freezes."

k4ot, 2021-12-13:
Try another power+USB cable. If it still makes the error, RMA

ambv, 2021-12-18:
Looks like a duplicate of fixed bug #903, let me know if you still have trouble after trying out [1.5.0b1](https://github.com/polyend/TrackerIssues/releases/tag/1.5.0b1).

closed 2021-12-18 by ambv