polyend/TrackerBetaTesting #1304: New MIDI CC: Pattern Selection

sunfallz, 2021-12-05:
Feature Request: New way to select Patterns

### Select Pattern via MIDI CC

I think it would be amazing to be able to simply select a pattern to jump to via incoming MIDI CC. 
This way when moving around on other sequencers they can sync together, or not if you don't want to. 
For example:
An external Sequencer has a simple 1-4 pattern chain event occurring, 
and I send midi cc to the Tracker from that sequencer to select any pattern of choice to play along.

For users such as myself who don't use Song Mode, pretty much ever. This would be an amazing feature to be able to move around sections in sync with, say, another Tracker, or any sequencer capable of sending MIDI CC's.

PEtajaja, 2021-12-06:
I really want this too as I've mentioned in a few other threads and especially if we're able to perform instant pattern changes rather than just sequential ones.

sunfallz, 2021-12-06:
agreed, this would be an awesome addition.