polyend/TrackerBetaTesting #1305: User changeable color theme

moshier1, 2021-12-06:
See title. The default theme is gross.

Jacknametrouble, 2021-12-06:
Thank you for your feedback.

closed 2021-12-06 by Jacknametrouble

bambrose, 2021-12-06:
I disagree about teh current theme (I really like it in fact)
But as someone who looks at typography all day I do find the innumerable font sizes a bit odd.
to my eyes it would look a lot better if there was basically one font and as few font sizes as possible..
In the preferences menu for example the headings are 1-2 pt bigger than the options within.. which looks weird.

trutypesounds, 2021-12-24:
> See title. The default theme is gross.

check the new one, you'll like it ahhahaha
[image]
[image]
[image]

trutypesounds, 2021-12-24:
user customizable theme yes please +1

moshier1, 2021-12-29:
> > See title. The default theme is gross.
> 
> check the new one, you'll like it ahhahaha [image] [image] [image]
I hate light themes and I love it. finally gonna b able to use the tacker outside come summer