polyend/TrackerBetaTesting #1306: Song mode - loop using multi consecutive slots

ketumatek, 2021-12-06:
hi; 
 it will be very nice to put this new feacture in song mode.

- have the possibility to loop blocks as 2 , 3, or more slots at the same time . to improve creativity i need to loop few slot to play on it .
for the moment i can loop only one slot, and it is very unuselfull always had to stop the sequencer go back play the few loops and doing that again. 

- as the same way i select the slots to make a copy or delete them i can select the slot i need and push shift+play to play them in a loop mode.
- please consider this request as something very important and really essentiel for live creativity.

tfspsound, 2021-12-07:
Agreed. Pressing Play twice quickly to make the the song position jump back is needlessly wearing out the Play button too.

colonel-blimp, 2022-01-21:
Good idea.  Similar issues: #598, #763