polyend/TrackerBetaTesting #1311: [1.4.0] Tracker freeze on particular attached project

Jacknametrouble, 2021-12-08:
name: Benjamin Palmer
email: bp@ncc.la
I have a repeatable error with this project file, which I have created on v 1.4.0

if I have the first track selected and go to Instrument Parameters and then use the destination keys to scroll down the list to Finetune and then the up arrow to go back up when it gets to Panning the device locks up and emits a quite loud high pitched tone out of the headphones - the keys are unresponsive I have to unplug it to

I have tested on other patches created on 1.4.0 or previous versions and can not reproduce this error on other project files, but I have been able to reproduce it on this file so I am attaching it as well as a video with the screen and the tone

Project where it happens:
[bp_tracker_error.zip](https://github.com/polyend/TrackerIssues/files/7675497/bp_tracker_error.zip)

Video:
https://user-images.githubusercontent.com/70910214/145197694-4914da16-04f1-472d-bb41-7a4f1448f2f1.MOV

thanks!

BP

MortalJS, 2021-12-08:
Bug confirmed.

ziej8asek, 2021-12-14:
It's the same issue as #1154 and #903, it will be fixed in next beta.

closed 2021-12-14 by ziej8asek