polyend/TrackerBetaTesting #1312: Undo in song mode not working if applyed on multiple tracks and pattern

CarloAvanti, 2021-12-08:
If you copy and paste multiple tracks and patterns in one time, on already used pattern and tracks, the undo command wont work… 

* How often does this bug happen? Every time

* Does this bug happen in previous firmware versions? Yes, at least the previous

* Does this bug happen in every project? It happened in different projects….

MortalJS, 2021-12-08:
Hi @CarloAvanti Can you add firmware version. Thanks

MortalJS, 2021-12-30:
Hi @CarloAvanti 

I have checked the problem you describe.
If you copied 4 paternas at once then the tracker will paste those 4 paternas into the grid. It will do it in one operation, but physically it will make 4 operations to copy 1 pattern.

To undo you need to press undo 4 times.

This is not a bug but a feature.

closed 2022-01-11 by MortalJS