polyend/TrackerBetaTesting #1318: Quick Way To Select MIDI Output Channel

SeanHub, 2021-12-13:
I use the tracker as the hub behind my music setup, I have multiple MIDI instruments and love performing live. In order to play a MIDI instrument through tracker you need to select instruments 51 to 66. Could the selection process be made quicker to aid live performance? Maybe shift + white buttons allows you to select the next numbers higher than 48 for example?

Jacknametrouble, 2021-12-13:
Sean, have you tried the Instrument function key hold + jog wheel shortcut for a fast MIDI instrument selection?

closed 2021-12-13 by Jacknametrouble

lucianthorr, 2022-05-14:
This might be a closed issue because this feature is new but I just realized you can hold Instrument and press the bottom-right grid button to jump down to Instrument 48.  From there, it's pretty quick to get to the midi channels.