polyend/TrackerBetaTesting #1326: [1.4.1b1] Sample recorder won't record if MIDI chord is playing

depaffect, 2021-12-18:
## Hi, sorry you had trouble!

### Steps to reproduce

Please describe the steps to reproduce the bug. Tell us what you noticed
happening and what you expected to happen instead.

If you have MIDI notes playing with the MIDI chord FX, and try to record that with the Sample Recorder (via line in connection), the recording will stop a split second after you start the pattern! When you stop the pattern, the recording will resume.

This doesn't happen if there is no chord FX on the MIDI notes.

### Attachments and links

Youtube video to show problem : https://youtu.be/7WG_3YL5uvY

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")
Every time. I restarted tracker, made new project... still happens. 

* Does this bug happen in previous firmware versions?
Unsure!

* Does this bug happen in every project?
Yes

egoshiner, 2021-12-29:
I thought this was intentional due to the live nature of playing chords on a more advanced synth/the irreplicatable aspects of modular. Tonight, I rendered stems/a pattern which was mainly composed of my TTSH Arp 26 being played using both the MIDI Ch Instrument, as well as trying the MIDI Chord effect. In the render, there is zero sound from the stereo Line-in and I just hear onboard samples, even isolating the individual track triggering the MIDI device.

From my perspective, this is more of an inadvertent feature-turned bug as it makes sense to rely on hard-set tracked samples being played with almost no generative aspect outside of the randomness they can permutate using. 

However, not even being able to step aside and record, because the MIDI Chord is playing, sounds too wrong.

MortalJS, 2021-12-30:
Bug confirmed.
@ziej8asek

Jacknametrouble, 2022-01-15:
Fixed in v1.5.0

closed 2022-01-15 by Jacknametrouble