polyend/TrackerBetaTesting #1328: [v1.5.0 beta 1] Samples normalized to zero after adding new sample

jamartinez4991, 2021-12-21:
Loading new samples will zero out the content of other samples in the project, and make the Tracker temporarily unstable. Pretty heartbreaking if you haven't saved in a bit. One of the handful of bugs I've encountered that directly effect the sample data or instrument slots. Bugs aside, the .pti update is a step in the right direction. Keep up the good work. 

### Steps to reproduce

Load several samples into a project, then load a relatively long sample. A smaller sample can trigger it, just not as frequently. May not happen every time, but having an active sequence seems to increase the odds. Having a sequence running will also make it apparent when several samples have been zeroed. 

Notes: 

The sample will retain the parameters of it's instrument slot, and its sample length, but will be completely devoid of content. 

The Tracker will freeze in certain circumstances when trying to navigate away from the problematic project, such as creating a new project or loading a different project. Upon rebooting, the silenced samples will be initialized instrument slots. 

Before this update, loading a new sample or replacing an existing one would commonly distort the playback speed of the other samples being played for a split second, presumably due to samples being reshuffled in a common buffer. Since this update, I don't hear that anymore and instead get this bug. My hunch is that there is some relation between the two.

### Attachments and links
There may be a video incoming.

* How often does this bug happen? (every time, often, "it only happened once")

Often; I've only played with the new firmware for about an hour, and it has occured at least five times. 

* Does this bug happen in previous firmware versions?

No, though I have experienced a similar bug where a sample + parameters will disappear after loading a new sample. Very rare.

* Does this bug happen in every project?

So far, yes. I've made a new project each time, and it will happen eventually. I tend to hot swap samples in existing slots regularly, and this rears its head pretty quickly.

Jacknametrouble, 2021-12-21:
Thank you for your feedback. Quick questions - how are you powering your Tracker? Does it happen on any power source or a particular one only?

Jacknametrouble, 2021-12-21:
Any chance we can get a video clip with the mentioned bug occurrence, please?

jamartinez4991, 2021-12-21:
I was powering the Tracker from its included power supply, from a wall outlet. I'm at work, but I may be able to supply a video this evening.

jamartinez4991, 2021-12-21:
Here's a video I just made of the issue. First new project after turning the Tracker on.

https://drive.google.com/file/d/1502E7tOrJfuF33lK0683FQW7bfbUGnm-/view?usp=drivesdk

Jacknametrouble, 2021-12-22:
Thank you so much for the very informative video. This will soon be fixed thanx to your help.

closed 2021-12-24 by ambv

Jacknametrouble, 2022-01-15:
Fixed in firmware v1.5.0