polyend/TrackerBetaTesting #1329: Option to Retain Midi Channel / Instrument Assigned to Track or Project

sunfallz, 2021-12-21:
(updated request with video) 
Request: Config Option to Assign and Retain the Midi Channel / Instrument per track, and have that track Retain that information when performing live or editing across tracks.

Current behavior: (Midi Channel or Instrument), I work almost exclusively with midi on the Tracker.

1- assign track 1 to midi ch 1, enter notes 
2 -move to track 3, assign to midi channel 3, enter notes
3- now, return to edit track 1, and you think you're editing the steps for your external synth on that same midi channel 1 you already set, but, it does not.  
All notes entered on empty steps of the Track 1, end up with Midi Channel 3.
*Notice tracks do not retain midi channels as already set.
-- And when you select midi ch 1 again to enter new steps on track 1, and THEN move to track 2 again, now your entering notes on track 2 with whatever you selected for track 1 in this endless back and forth cycle...and now your external synths are confused and notes are firing off to synths you didn't want them to.
*Essentially retaining last selected Instrument across track selection.

It would be amazing to be able to set and forget a tracks midi channel without having to REASSIGN each steps midi channel / Instrument when you move around the pattern. 
Cheers!

https://user-images.githubusercontent.com/77541692/146856833-ae4b54f9-6aaf-4dab-98c9-a1bf4865dd4a.mov

piotr3500, 2021-12-21:
This is how the Tracker works. Having "unchained" steps is its huge advantage, although in this case might be a bit of pain.
I hope you can get used to it.

closed 2021-12-21 by piotr3500

sunfallz, 2021-12-21:
I realize it's how it 'works'. that's why it's a request, not a bug. 
instruments, such as sequencers, don't "need" fm radios. but here we are.
it's a constant struggle to simply return to a track and continue sequencing where you left off. but I get it. it's the tracker charm. 
still my current favorite sequencer. 
cheers.

sunfallz, 2021-12-22:
my old request of the same idea drew interest & comments from quite a few tracker owners as well, and on other forums. seems like a no brainer addition to any new sequencer. would take that tracker flow up a notch. 

so basically, you're saying this can't be functionally added. as in its an impossible code addition?
just trying to get a feel for the response. "get used to it".

still, current favorite sequencer.
cheers.

piotr3500, 2021-12-23:
I meant that if we implement your idea, then the former UX is changed.
Instruments/Channels are independent of tracks. This is one of the bigest strengths of trackrs. 
And, once you get used to it, you can be very fast.

Also, glad you like it :)